vBJohannes/SolderingIron #18: New Hardware Version

debaal, 2015-02-03:
- Check collision of parts (especially overlapping)
- Change pin for PWM of soldering iron to a pin connected to an 8-bit timer output
- Simplify Layout and create a THT version for users with no SMD experience.
- Add an optional FTDI for debugging
- Add an optional ESP8266
- More Buttons for Presets

kartenkarsten, 2015-02-09:
FTDI with bitbang mode for (inital) programming would be nice.

yschroeder, 2015-02-09:
This is not a dev board but a soldering station. I think a user of this project will likely get a proprogrammed avr and the ability to reprogram it via a dedicated programmer.

kartenkarsten, 2015-02-09:
Sure - it depends if there will be a shop selling this hardware - or if a customer has to order all parts by himself. Not every one soldering has AVR experience.

yschroeder, 2015-02-09:
Yes, we have to define who is the target audience for this project and how it can be delivered to a interested user. Depending on that we are able to make assumptions on what the user will be able to handle himself and what must be done before he gets a kit or however this project is shipped.

debaal, 2015-02-09:
Maybe we offer packages (for a slightly higher price) with preassembled and tested critical parts like the AVR, FETs, SMD Cs/Rs. This way we can flash and check the AVR. All bigger parts (LM324, LDO, THT parts) can be assembled by the user of course.

closed 2015-04-29 by debaal