vBJohannes/SolderingIron #19: Hall Sensor Integration

debaal, 2015-02-03:
To make the soldering iron realize that it is at rest a hall sensor is used that is triggered, when the iron is resting. The sensor needs to be interfaced, and tested for different orientation and magnets.

yschroeder, 2015-02-09:
Is this really due to the shipping of the project to the supporters? Is it part of the package we announced to them or is it more a feature we want to have in 1.0?

debaal, 2015-02-09:
It is a feature, that should be supported regardless of whether or not the user wants to use it. We need another channel of the LM324 for this anyway. The AVR should check the pin in any case, so the default is to make it as easy as possible for the user to use the feature or leave it out. This makes it critical, that the LM324 has the right resistors connected. And that what I had in mind when opening up this issue.

yschroeder, 2015-02-09:
This does not anwser my question in any way...

I am currently messing around with the milestones and I just wanted to know if this is due to the first milestone when we give out the ten ordered boards together with the ordered parts because the early users expect this to be working or if this is just a feature that is supposed to work when everything is polished and will be put to work by the early users if they want that feature later.

yschroeder, 2015-02-09:
If this needs hardware that is not present on the current boards though, we should mark this as a feature for 1.0 because it needs a new board layout.

oni303, 2015-02-09:
If you guys would have written a tenth of the text you wrote in lines of code this and many other features would have been finished by now.

kenog, 2015-02-10:
Full ACK, @oni303 :)

kartenkarsten, 2015-02-10:
As far as I remember we don't even need an opamp - using adc readings it should be possible to detect it.

debaal, 2015-02-11:
nope. it works without an op amp. but we should change some things (#60) to supply it with 5V.

debaal, 2015-02-11:
Is now working and is connected to ADC7.

closed 2015-02-11 by debaal

yschroeder, 2015-02-11:
Have you changed the pin? Should this be noted on the hardware changes page?