vBJohannes/SolderingIron #26: Check interaction of heating current with thermocouple

yschroeder, 2015-02-04:
As the soldering tip has only three connections and the thermocouple and the heater share a common ground we should check if there is happening anything funny to the temperature reading when the heater is powered.

If yes, we need a mechanism that notifies the adc library that the power is off and the temperature can now be measured.

By thinking about it there shouldn't be any interference as the thermocouple is a voltage source and the circuits from it and the heater are separated, but let's watch out for this!

yschroeder, 2015-02-04:
The interaction is resolved on paper @oni303 has the documentation to integrate it into the PID controller

yschroeder, 2015-02-04:
The PID controller should not use the ADC interrupt directly. The ADC library will provide a callback when the tip temperature conversion is complete. The tip temperature can then be read from a global variable.
This global variable will also be used by the display controller to show the current temperature to the user.

oni303, 2015-02-04:
see #30

closed 2015-02-04 by debaal