vBJohannes/SolderingIron #31: Another Display

debaal, 2015-02-04:
The widespread HD44780 should be supported. The problem is, that it needs at least 6 pins to make it work. This could be done by using Port B. That should be free, when the current display is removed.

Maybe also the DogL displays might be an interesting option.

We could keep that as an option, so the user can decide, what display he wants. We might have to make a pin header to stack other display shields onto the base plate.

yschroeder, 2015-02-04:
Does this need to be that versatile? If you use real LCDs you need a lot more software for menus and what not. I personally like the simplicity of the current user interface.

oni303, 2015-02-04:
Lets be focused on getting this version working first. Afterwards you are free come up with fancy ideas.

FATANIKA, 2020-09-14:
Matap semua bisa..bisnis

FATANIKA, 2020-09-14:
Aku butuh akses.yg kuat

FATANIKA, 2020-09-14:
[image]
Go berkah no riba

FATANIKA, 2020-09-14:
[image]
[image]

FATANIKA, 2020-09-14:
[image]

FATANIKA, 2020-09-14:
@pakfatanika@gmail.com

FATANIKA, 2020-09-14:
[image]
[image]