vBJohannes/SolderingIron #34: Default PID values

debaal, 2015-02-04:
PID values should be defined as a global variable to go easy on the EEPROM. Also there should be default values which should be used in case there are no values written to the EEPROM or the EEPROM breaks.

Custom values can be written to EEPROM (see #33) and will be used then.

oni303, 2015-02-04:
lets not use variables for static values like, lets take defines

debaal, 2015-02-04:
But you might not want to read the EEPROM everytime you run the calculation.

yschroeder, 2015-02-05:
1. define default values in config.h
2. initialize P, I ,D variables with these values
3. check if other values are present in the eeprom
4. update the P, I, D variables if eeprom has data
5. ...
6. profit

oni303, 2015-02-12:
DONE

closed 2015-02-12 by oni303