vBJohannes/SolderingIron #43: No tip detection

yschroeder, 2015-02-05:
We can implement a reliable detection if no tip is detected by adding a resistor to the op amp input that pulls it to a defined voltage (either GND or VCC, more on that later). This way the op amp output would go into an extreme which signals that no tip is attached.

I think pulling it up to VCC is the best option, this way it is possible to even detect a short circuit on the sensor pin, when it is connected to GND somehow.

An ADC value of 1023 signals a missing tip then, an ADC value of 0 (or close by) signals some other defect of the tip (shorted sensor pin).

By pulling up the op amp input we will change the calibration curve that we sampled earlier. This should be resampled then.

Someone should validate that this works and we should find a reasonable resistor value as pull up. I think 100k may be too little, but let's experiment on that.

kartenkarsten, 2015-02-05:
Maybe this is the reason of 100k pull down on the other board.

yschroeder, 2015-02-05:
Yes, thats what I think, too.
But when experimenting with that we have seen worse output from the op amp (as I remember), so maybe this should be change to pull up and/or higher value.

debaal, 2015-02-05:
Maybe an even higher resistor? 1M? 10M?

oni303, 2015-02-05:
Maybe I found a solution to solve the problem without hardware modifications. See the Controller PR.

yschroeder, 2015-02-06:
After testing the detection can be easily implemented:
The op amp saturates at about 3.7 V when no tip is connected, even without any pull up resistor. The corresponding temperature would be 450°C.
The resulting configuration could be as follows.
Allow a maximum temperature for the user to select of 400°C. Above 440° the temperature is detected as errornous which may be due to a not connected tip and a flag is set. This allows a margin of 40°C for the PID controller. When the flag is set, the FET for heating the tip is not allowed to be activated! When the tip temperature sinks under 400° the flag is cleared and the tip can be heated again.

yschroeder, 2015-02-09:
Removed the hardware label as this can be done in software only.

oni303, 2015-02-12:
I found a solution which is more reliable than reaction only to temperature.
It measures the differences between the powered tip and the not powered tip. If there is no difference between them, no tip is connected.

yschroeder, 2015-02-17:
implemented in #61

closed 2015-02-17 by yschroeder