vBJohannes/SolderingIron #45: return value of controller init

kartenkarsten, 2015-02-06:
the return value of controller init should be a function pointer to be triggered after a new temperature was read

oni303, 2015-02-09:
does not make sense to me.
In my opinion the main will all controller_init, the controller than calls everything it needs to work.

yschroeder, 2015-02-09:
The point is that the tip can be used for different things than just temperature control. The controller should not inititialize it, this is a task for the main-routine. It just needs to hand back its function pointer that should be called whenever a new temperature reading is available. This way the main-routine can hand that pointer over to the tip. This cleans up the code.

oni303, 2015-02-09:
I am not sure what you are planning to do with your tip but I will use mine to solder and to do so it needs to be heated.
Using the controller without a properly initialized tip will destroy the tip. Therefore, the controller will initialize the tip .

oni303, 2015-02-12:
as described above this change will no be made

closed 2015-02-12 by oni303