vBJohannes/SolderingIron #47: Cold Junction Compensation for the thermocouple

yschroeder, 2015-02-07:
I was informed (and I should also know this from lectures...) that we have to be careful with the readings from the thermocouple regarding its cold junction temperature.

Small excursion to the world of thermocouples:
A thermocouple only measures a differential temperature. The voltage at its output is proportional to the temperature _difference_ between the coupling point and the end that is connected to the measurement circuit.

Back to our soldering iron:
This may lead to complications with our design. We have calibrated the tip temperature already at what must be something like room temperature. If you hold the soldering iron in your hand then you will likely change (increase) the temperature at the cold junction. This means that we will measure a smaller temperature and therefore the tip is hotter than expected. Also, what happens when the soldering iron is on for longer and the grip gets warm? This will also decrease the measured temperature.

I am not sure how this is solved in the original design, but maybe this is the reason why the complete grip is made from metal. It may dissipate heat from the cold junction.

If the heat dissipation from the tip to the end with the connector can be ignored, then the error in the temperature will be around 10 Kelvin from the warm hand of the user. This is acceptable. However if the heat from the tip heats the connection significantly, then the tip temperature will be much higher than the measured temperature.

We can test how the original iron behaves by cooling the end with the connection down while it is heated to some constant temperature. If they don't take care of this, then the display should wrongfully indicate a rise in temperature.

kartenkarsten, 2015-02-08:
If we want a good result we could measure the handle temperature. But I think it should be sufficent if we start with the assumption of 23°C degree handle Temperature.

yschroeder, 2015-02-10:
We have tested if we can disturb the tip by cooling down the end of tip with the audio jack. We could neither observe any changes in the tip temperature (measured with thermal imaging camera) nor in the temperature measured by the soldering station. We tested this on both, our station and the original one.