vBJohannes/SolderingIron #52: what is the current Hardware version?

kartenkarsten, 2015-02-09:
documenting the Pinouts I wondered which Hardware verison is connected to witch Milestone. Am I right that milestone *running verion* is ```HW_VERSION=2``` and our current PCB has ```HW_VERSION=1``` ?

yschroeder, 2015-02-09:
In my opinion it is as you stated, but we will discuss tomorrow if we give the current hardware to the users who ordered a kit or if the revised and fixed version will be handed out for real use.

The changes made to the boards from the first Itead order to be ready to use (added resistors and what not) should be documented anyway.

kartenkarsten, 2015-02-09:
started a wiki section to document [hardware fixes](https://github.com/debaal/SolderingIron/wiki/Soldering-Tipps#hardware-fixes)

yschroeder, 2015-02-09:
:+1: We should add photos (maybe shot though the microscope) to clarify how to apply the fixes

debaal, 2015-02-09:
In v1.brd and v1.sch all the fixes and ideas are included, which we had while developing. Since the original layout that was ordered is not working the way it should. We had to make two little adjustments (see wiki).
Since these adjustments should be done on all pcbs already there no real second version right now.

If the first soldering irons are fully up and running I'd like to discuss a release of a fixed first version (eg v1.1).

Right now the layout of the pcb and the one online should include all the fixes and I would still call them v1.

kartenkarsten, 2015-02-09:
the problem I see is we have a define ```HW_VERSION``` I would like to use this to reference to specific changes (its a Integer). We should sync it some how with the github release feature / git tags. Any Idea?

yschroeder, 2015-02-09:
Ok, so as I see it, the hardware version that is currently used for development is some pre 1.0 version which needs no actual hardware version in the code as it will be made compatible to 1.0 by adding the missing hardware. In that case the wiki should mention this board as beta version without an explicit reference to the `HW_VERSION` define. It may be noted that the changes documented in the wiki will make the beta board compatible to version 1.0.

The define may be changed to `10`. This would represent version 1.0 and a later version 1.1 can be easily implemented as `11`.

debaal, 2015-04-28:
We changed a lot since the first feedback. So the current version of hardware is now 1.2 and that will be most likely the one for the next release.
So software should be developed for that version. Some adaptions were already made.

kartenkarsten, 2015-10-19:
Do we want to delete all gerber files and introduce releases where these files are included?