vBJohannes/SolderingIron #53: Build Environment

yschroeder, 2015-02-09:
The wiki currently states that we build this on linux. I see no reason why this project can be built on Windows with WinAVR. It supplies `make` and `avrdude` for Windows and should work fine.

Should we support (and test) that, too? And what's with Atmel Studio? That may work as well...

oni303, 2015-02-09:
Well if you like you can try to do so, I will not.
We have to make sure that both version will be supported in future.

kartenkarsten, 2015-02-10:
If we want to become a bigger user base a windows tutorial would be good - but more work. I'm ok with a linux only version, but we could add precompiled hex-files to each release to ensure everyone can at least program the hardware.