vBJohannes/SolderingIron #55: Use a logical numbering for the Resistors and capacitors on the board

yschroeder, 2015-02-09:
I know this one is not easy to pull off with Eagle, but the numbering is odd in some places. For example the numbers for the current limiting resistors on the display are messed up. I wanted to reference then in the FAQ but it is not possible to just say R4-11 or something like that.

On the other hand it is not a good idea to change the numbering all the time as it will kill compatibilty between hardware versions and may mess things up even further.

We should get this right as early as possible to reduce chaos.

debaal, 2015-02-10:
how about assigning different ranges to different areas like R100-R199 are for current limiting resistors for the display, R200-R299 are all for the ones on the display's low side around the fets.

yschroeder, 2015-02-10:
That may confuse people who think every number in the whole range is assigned and make them think we have hundreds of resistors. Also it uses much space as a silkscreen. But I think I like the idea after all.

debaal, 2015-04-30:
Fixed..

closed 2015-04-30 by debaal