vBJohannes/SolderingIron #58: Add an option to change temperature to Fahrenheit

yschroeder, 2015-02-10:
There may be users who prefer to display the temperature in Fahrenheit. There should be a way to change the displayed temperature unit.

oni303, 2015-02-11:
I do not support units based on the coldest day in 1708 in Gdańsk. Is like taking a random foot of some random king to measure distances. 
Also we do not have the digit to indicate the unit.

debaal, 2015-02-11:
i actually have a solution at hand and could push it but oni and kenog kind of disapprove of that.

yschroeder, 2015-02-11:
Well, if we do not support Fahrenheit we sould also not support Celsius but only Kelvin as Celsius is defined by water temperature which is also pointless as we are dealing with tin melting temperatures...

I am not sure how soldering temperatures are handled in america, maybe it is also valid there to use Celsius but as you can set the temperature scale on other soldering irons, too, I thought it might be a nice option.

oni303, 2015-02-11:
freezing water is a much more reliable base than that day in 1708 or
freezing " mixture of ice, water, and ammonium chloride, a salt, at a
1:1:1 ratio"[1]

But I am ok with having "funny unit support" with temperature scales like:
* Rankine (valid unit in the USA)
* Delisle (valid at some time in the past in Russia)
* Réaumur (valid at some time in the past in Europe)
* Newton  (valid nowhere, never)
* Rømer   (valid nowhere, never)
* Fahrenheit (well unit only in the USA)

[1] http://en.wikipedia.org/wiki/Fahrenheit

yschroeder, 2015-02-11:
Ok, then it is set.
We need a define in `config.h` that allows all of the units @oni303 named. But we should make it as transparent as possible. all internals will still use celsius, only the display tells otherwise and the buttons should always count one up and down in the selected unit, not 1°C. That would cause too much confusion.

debaal, 2015-02-11:
I'm on it..

debaal, 2015-02-11:
done. (commit 6d11af117e2b8ceee2420c8064a899c11214336a)

please use display_temperature(int16_t number) from now on instead of display_number(int16_t number) when displaying a temperature.

oni303, 2015-02-11:
This is not done yet.
If you want to support these units you have to build the same conversion for the buttons too.
Otherwise the associated commit should be reverted.

debaal, 2015-02-12:
2 thoughts for that:

- since we display a desired temperature when you hit a button, this temperature will be translated to (let's say) Newton as well. So in the background the software tries to heat to 1°C more, but will display your new desired temperature in newton. You might not be able to hit exactly for example 150°N, but only 149 and 151. But I could live with that. This way people see at least the units they are used to. 

- the other way would be calculating a different temperature right when reading ADC0. So a voltage is translated directly to Fahrenheit or Kelvin). Since we are just working with numbers (not real temperatures) that should not be an issue heating from 100°C to 180°C or from 250°F to 450°F.
We might to tune PID values a little and translate MAX_TEMP to other scales. But that's it. This way a user can always navigate to any specific temperature in his favourite unit. 

I think both ways are possible. The first one seems easier and the second one more independent from the actual temperature on a specific scale.