vBJohannes/SolderingIron #59: Tip safety in hardware

yschroeder, 2015-02-11:
You may have already heard about it: There is alredy one dead tip from experimenting with the PID controller.

If powered from a beefy 12V supply the tip heats up until it glows bright orange in a few seconds. If something goes wrong it dies in seconds!

Therefore we should implement some mechanic to disable the heating when an absolute maximum temperature (like 550°C) is reached. We already discussed some ways to implement it.

The best option we could come up with so far is to use another channel of the op amp and let its output pull the reset line of the AVR when the temperature reaches the critical point. Reseting the AVR will instantly disable the heating. After that the AVR enters its reset interrupt and stays there indefinitely while printing an error on the display. This condition can only be changed by power-cycling the whole soldering station.

There is just one problem: The op amp will have to measure the tip temperature when it is heated which itself results in huge false tempature readings. The read temperatures are even dependant on the resistance of the cable to the tip as there is a voltage over that wire while the current is flowing. Maybe we can find an upper bound for that voltage and let it trigger when it exceeds this voltage.

debaal, 2015-02-11:
we had an idea to make sure the tip will not over heat. hope this helps.
[image]

yschroeder, 2015-02-11:
I think there are a few issues with this solution:

First you use the output of the first op amp 1A as input for the safety stage. I am not sure if the op amp 1A saturates if the tip is heated. If it does, you will not be able to measure the temperature while heating which is critical for proper operation.

We should use the fourth op amp 1D to make a second amplifier for the thermo voltage but with a smaller gain to make sure it does not saturate when the heating is on. Connect that output to 1B. 1B should be used as a voltage comparator, by adjusting its reference voltage we can control the temperature when the system shuts down.

The voltage output by the schmitt trigger may be too low. The datasheet states that the reset may already trigger if the voltage is below 0.9*VCC. The op amps saturation voltage is lower and can trigger the reset if we are unlucky.

The resistor after the schmitt trigger may be too big. The voltage on the reset pin must be smaller than 0.2*Vcc as stated in the datasheet. The next FET is not needed. I think we can be sure that the AVR resets properly if the reset pin is asserted. There is no need to make extra sure that the pin that controls the power is turned off.

oni303, 2015-02-11:
Actually the AVR is not supposed to be resetted.
We just withhold the FET from heating the tip.

yschroeder, 2015-02-11:
Oh yes, my bad. I somehow thought PC1 is the reset pin.

I am ok with only disable the heating and notify the AVR via PC1 that the satefy feature was triggered.

But I still don't know how you plan to set the voltage at which the safety feature is triggered.

debaal, 2015-02-11:
the idea is that the parts on the right triggers if the voltage of 1A is too high for too long.
because that means either it is heating non-stop or too hot

yschroeder, 2015-02-11:
Ok, but shouldn't 1B then be an integrating op amp? That way it would track how much of the time the tip is heated.

debaal, 2015-02-11:
could be implemented as an integrator as well. this way the iron will shut down if the filtered voltage of heating and measuring is too high. this happens if you heat a hot iron a lot.
or you build an integrator and check how much the iron is heated. but this does not compensate for cooling the tip a lot (e.g. heating large objects or sticking the iron in water).

yschroeder, 2015-02-11:
have you already tested your design?

debaal, 2015-02-11:
not at all. it's nothing but an idea this far.

debaal, 2015-04-28:
I think it's done. Tested on a breadboard and implemented in hardware in v1.2 (see bc359611b22779060904d67200dcd036d9aafa12)

closed 2015-04-28 by debaal