vBJohannes/SolderingIron #68: USB power delivery

oni303, 2015-02-14:
integrate Microchip UPD1002 and USB type connector for power supply.

yschroeder, 2015-02-14:
I didn't know that this standard existed until your issue. Are there compatible devices available? Can I use my yellow usb port on my thinkpad to power a soldering iron? There seem to be different profiles and not all of them are suitable for a soldering iron. 

Also the package of that chip is not easy to solder. Maybe we should build a hot air gun before...

oni303, 2015-02-14:
No the yellow port is suitable for this. Actually there is on power
supply by now and no cables, no normal USB cables are not suitable to 100W.
There is a consumer profile for 12V 5A.

+1 for the hot air gun ;)

oni303, 2015-03-10:
good news!!!

Yesterday Apple presented the new Mac Book which is charged using usb power delivery over an usb c connector.

I think we will see usb c power supplies from other supplierers soon.