vBJohannes/SolderingIron #73: Cable

debaal, 2015-04-30:
We also still need some wire to run from the soldering station to the tip.
Right now it is a bundle of 3 1mm^2 wires and 2 0.14mm^2 wires.
Do we sleeve them? Duct tape? Put a spiral* around it?

*http://www.reichelt.de/Spiralschlauch/SPIRAL-08-10/3/index.html?&ACTION=3&LA=2&ARTICLE=50452&GROUPID=5714&artnr=SPIRAL+08-10

oni303, 2015-04-30:
As we discussed earlier I think we can use some sort of 10 wire cable and use multiple wires for power and ground.

debaal, 2015-04-30:
We got a solution now:
LIYY 7x 0,25mm^2 (2 GND, 2 12V, 1 Hall, 1 Temp, 1 VCC)
Available at Conrad. We ordered a sample of 2m.
1.5m should be sufficient in the end.

oni303, 2015-04-30:
Sounds good, let's see how good the handling is with this cable

debaal, 2015-06-05:
Cable handles fine. We used a this one: Datenleitung UNITRONIC® LiYY 7 x 0.25 mm²

closed 2015-06-05 by debaal